whitequark/solvespace #236: Assign to style

Leichtbier, 2016-08-06:
If I import a dxf file than all geometry has the style s100-@dxf. If I select for example a line an use:

Assign to style > none

it works as expected. It also works for a smaler selction. However, I selct all geometry at once and try to do the same operation it fails.

The used file:
[bottom.zip](https://github.com/whitequark/solvespace/files/405053/bottom.zip)

Best regards

Leichtbier

Evil-Spirit, 2016-08-07:
@Leichtbier, you can not assign styles for points, and this is why you can't assign style to selection which involves some points. But you still can use "select edge chain" option by right clicking some entity.

Leichtbier, 2016-08-07:
Maybe it's appropriate to notify the user if the selection is invalid for the desired operation. 

Despite I've the same proplem with "select edge chain" > "remove assigned style".
Although there is only one chain there are still some lines left with the style s100-@dxf.

Best regards

Leichtbier

Symbian9, 2016-08-07:
Think, its can be solved by automatically hide "Assign to Style" option in contextual menu if in selection presented at least one point

whitequark, 2016-08-13:
Figured out the problem. Selecting over 32 of anything causes the rest of the selection to be ignored in mass operations. Fortunately, easy to fix.

whitequark, 2016-08-13:
Fixed in master and 2.x.

closed 2016-08-13 by whitequark