whitequark/solvespace #237: OBJ export improvements

Evil-Spirit, 2016-08-14:
I've made some improvemnts for OBJ export
https://github.com/Evil-Spirit/solvespace/commit/3c385f7b3819b7bff21eaf94af87a6079cacc3a9
* normals exported
* material colors exported

Symbian9, 2016-08-19:
Its strange for me that in exported STL mesh all faces are separated, but in OBJ faces connected as it must be. So, is it an STL export issue or STL format cant include stitched faces?

Evil-Spirit, 2016-08-19:
There is some code for stitching it. But this breaks "smooth groups" and normals, so I get rid of this inside this improvement. We still can add stitching but have to compare coordinates and normals at the same time to preserve "smooth groups". STL files don't support indexed mesh representation, so each triangle saved separately.

Symbian9, 2016-08-20:
Ok, I hope you fix it ))
Also, can you give names of all files that include mesh export code (for OBJ). Will try write AC3D export test

whitequark, 2016-10-11:
Merged in master.

@Symbian9 try starting from STL format. Look for the `ExportMeshAsStlTo` function and `if(FilenameHasExtension(filename, ".stl")) {` inside `ExportMeshTo` function.

closed 2016-10-11 by whitequark