whitequark/solvespace #239: Offset geometry and sketch variables

wavexx, 2016-08-20:
I wish there would be a way to offset (at least planar) geometry, reference and evaluate variables into a sketch. They go hand-in-hand. They make a huge difference for machining.

I'm generally expecting to be able to design with optimal dimensions in mind, and add a final offset in order to compensate for the tolerances where needed. This is machine and material specific though, so I'd like to abstract it away from the piece.

In many cases it's generally easier to do this by offsetting the final geometry of the piece, where you offset the sketch by a defined amount (very common for laser cutting and simple surfaces). I'm using inkscape "dynamic" offset for this as a last step (I'm not kidding...).

In other cases you have to take tolerances into account when adding constraints, but you'd have to reference some sort of variable to reuse over and over in the constrains which are then evaluated symbolically.

Maybe it's a bit over-arching for a single request ;), but it would be nice to:

- introduce variables by defining them sort of like text comments of the form "name = value" in the sketch.
- introduce simple expression evaluation for the constraint values. Showing the final value is perfect, but the expression need to be stored un-evaluated, of course.
- re-evaluate the constraints when variable are changed

I can actually help for those. Doing this would already allow me to use solvespace almost exclusively to the production stage, as I could go by without offsetting already by just doing this at a lower level.

I'm not so sure about actually implementing the offsetting operator, as I never gave solvespace's kernel a big look to see how complex would it be.

whitequark, 2016-08-20:
Yep, variables are a widely desired feature. They are fairly high priority, and there's already an issue for them: https://github.com/whitequark/solvespace/issues/194.

As for offsetting for laser cutting, this can already be done when exporting plane sketches. This can be set in the configuration. The results produced by it are similar to what Inkscape does. Offsetting is done on the piecewise linear representation. If you want offsetting that works on exact curves instead, please file a separate issue for that.

closed 2016-08-20 by whitequark

wavexx, 2016-08-20:
Interestingly I couldn't find it by searching for "variables". Sorry for the duplicate.

As for offsetting, a value for exporting is nice, but being able to define offsetting internally is needed anyway in many cases.

whitequark, 2016-08-20:
> Interestingly I couldn't find it by searching for "variables". Sorry for the duplicate.

Yeah, I've adjusted the title of that issue because of this. Should be fine now.

> As for offsetting, a value for exporting is nice, but being able to define offsetting internally is needed anyway in many cases.

I'm unaware of any easy way to do it on exact curves, so don't get your hopes up.

wavexx, 2016-08-20:
I know it's not easy ;)

whitequark, 2016-08-20:
Well, maybe we can use the same thing inkscape does. Though I found its results quite unsatisfying in the past, personally.

wavexx, 2016-08-20:
On Sat, Aug 20 2016, whitequark <notifications@github.com> wrote:
> Well, maybe we can use the same thing inkscape does. Though I found its results quite unsatisfying in the past, personally.

I'm not sure about what inkscape uses, but I used cgal's 2d skeleton
offetting for another project in the past and it was fine. But it's also
not exact for curves, of course.