whitequark/solvespace #240: Make all groups as assemblies

Evil-Spirit, 2016-10-23:
Some people already created files like modelisation.slvs and this can be useful for fast migration
https://github.com/Evil-Spirit/solvespace/commit/6e9b115f7449e8583a6a8d7072e2d68d508d29c7

whitequark, 2016-10-23:
1) this is not actually faster.
2) modelisation.slvs will never become faster even if we implement this in an optimized way.
3) this is adding too much junk to the menus.

Evil-Spirit, 2016-10-23:
1) this actually pressing one button instead of changing each of >20 groups
2) this farster for me 40s -> 10s on loading. I don't see any reason why we shouldn't do files like this woriking in realtime.

whitequark, 2016-10-23:
Did you generate *all* groups as assemblies? Or did you actually fix the boolean operation errors and switched back the groups that failed them? After I did the latter, it made no difference.

Evil-Spirit, 2016-10-23:
@whitequark.
Ofc all the groups. Because asm is not for fixing booleans, this is for constructions like modelisation.slvs which actually asseblies but it is hard to build it with separate parts.

whitequark, 2016-10-25:
> Because asm is not for fixing booleans, this is for constructions like modelisation.slvs which actually asseblies but it is hard to build it using separate parts.

But modelisation.slvs is not a valid assembly, it has parts that penetrate into each other. That's exactly the problem here.

Evil-Spirit, 2016-10-25:
@whitequark,
Ofcourse, to build real shopdrawing assembly of wooden constructions SolveSpace needs to have tree of groups with boolean operations. For the current case, some advanced wooden details can be built like separated parts. Anyway, this approach still much simplier than making fully part-by-part and applicable for most of details of construction.  SolveSpace can be much better than SketchUP if problems with excessive generation (regeneration of entities of ALL groups instead of just solving from current while dragging)

whitequark, 2016-10-25:
My point is that existing union sketches will result in self-intersecting meshes if they are blindly converted into assemblies. I see no use in making it easier to achieve a broken result.

Evil-Spirit, 2016-10-25:
@whitequark, the problem is general. You can assemble imported parts the same way and get the same broken result.

whitequark, 2016-10-25:
We are talking about existing sketches here. Linked parts already come as assemblies, and so if they intersect, these sketches are already broken. Your feature takes perfectly valid sketches with unions, and breaks them. Correctness comes before performance.

whitequark, 2016-11-02:
Closing because a feature that takes non-self-interfering sketches and makes them self-interfering, without any regard, is a bad feature.

closed 2016-11-02 by whitequark