whitequark/solvespace #241: default filename at export

tgirod, 2016-11-04:
In version 2.1, exported files are called `untitled.EXT` by default. I think it would be more convenient to call them after the name of the exported file - which is the behaviour of most applications, I think.

That's a very low hanging fruit, I might be able to do it if you folks are ok and I find the time.

whitequark, 2016-11-04:
This is actually deliberate. Especially when exporting 2d views, often you want to export more than one, and this "untitled" suggests that the name should be changed to something descriptive.

tgirod, 2016-11-04:
IMHO it is better not to make any assumptions about how the user uses a function, and follow a widely accepted, unsurprising behaviour.

If we are to make assumptions, I think it makes sense to have the exported filename as a prefix to all the 2D views exported. So the export behaviour could be altered to `EXPORTED.EXT`, with `EXPORTED` selected, so the user can suppress the filename or quickly add something after.

whitequark, 2016-11-04:
> So the export behaviour could be altered to EXPORTED.EXT, with EXPORTED selected, so the user can suppress the filename or quickly add something after.

Makes sense. I'll implement this.

whitequark, 2016-11-17:
Fixed in master.

closed 2016-11-17 by whitequark