whitequark/solvespace #242: Some small forgotten commits

Evil-Spirit, 2016-12-22:
Automatically mark newly-created redundant constraint as reference
https://github.com/Evil-Spirit/solvespace/commit/9e6d030458bea9d3c1adafb85780ca12260daca5

Do Schedule GenerateALL
https://github.com/Evil-Spirit/solvespace/commit/c35d2787941b4849a811ea6be24138411950dc84

Some hot BSP function optimized
https://github.com/Evil-Spirit/solvespace/commit/ad904667d1966daa148035202469479a66ad5253

whitequark, 2016-12-22:
@Evil-Spirit see comments in Evil-Spirit@c35d278

Evil-Spirit, 2016-12-22:
@whitequark, oh, ye. I forgot about this.

whitequark, 2016-12-22:
> Automatically mark newly-created redundant constraint as reference
> Evil-Spirit@9e6d030

This is not small, this pulls in a huge amount of code. And it doesn't apply on master.

Evil-Spirit, 2016-12-23:
@whitequark,
Actually, I just want to hear you opinion if we need to merge it. This need to be rebased and fixed since we made some things different way.

whitequark, 2016-12-23:
@Evil-Spirit We can populate dofDirty right in AddConstraint, right? If we can do that then there is not much more work involved than in regular generation, and yes we can do that. If not I am afraid this will slow things down too much.

whitequark, 2016-12-30:
Both remaining commits need your action

Evil-Spirit, 2016-12-30:
@whitequark, OK.

Evil-Spirit, 2017-01-11:
@whitequark, remake with fast redundant test https://github.com/Evil-Spirit/solvespace-master/commit/722c1dd914da2495c8335c2811ea0dfe1f5d43c0

Evil-Spirit, 2017-01-11:
3rd is not actual because of allocations

whitequark, 2017-01-12:
Merged

Evil-Spirit, 2017-01-14:
@whitequark, fixed here https://github.com/Evil-Spirit/solvespace-master/commit/1ce66084fac71e7eb8e63098a899d69509363d63

to reproduce the bug, draw some rect and make translate group. then constrain point of some rect to line of the different rect (PT_ON_LINE).

whitequark, 2017-01-16:
Merged

closed 2017-01-16 by whitequark